periph/devices #6: Interested in adding support for Adafruit's NeoTrellis, could use some guidance though.

different55, 2021-02-18:
I already use periph to control a ssd1306 and it's been working great! Recently got one of Adafruit's NeoTrellis boards to add to the project and I wanted to try to contribute support for that. The NeoTrellis is kind of an odd duck though. It's actually just one of their "Seesaw" boards pre-configured for 16 WS2812s and 16 push buttons. And the Seesaw itself is "just" an atsamd09 with firmware to break out its GPIOs over I2C.

What would be the best way to approach this? Looking at Adafruit's Python driver for the device, it mostly just wraps the existing Seesaw libraries for their button and LED support. There's nothing like that here, so would it be best to start by implementing support for the Seesaw and adding a wrapper the same way that Adafruit does? Or maybe just implementing only what's needed for this exact chip I'm interested in?

maruel, 2021-02-19:
It's always tricky for boards with multiple chipsets and specific firmware.

For drivers where you effectively support a proprietary firmware (here it's https://github.com/adafruit/seesaw), I'd prefer to put them under a new directory "adafruit", so the package path could be `adafruit/seesaw`. Then you can create another driver that compose over it for the NeoTrellis, so `adafruit/neotrellis` that would embed a seesaw. See a precedent with https://pkg.go.dev/periph.io/x/devices/v3/lepton#Dev but feel free to completely hide it if preferable.

In short, try to decompose as much as it makes sense, but not "too much". According to 
https://learn.adafruit.com/adafruit-seesaw-atsamd09-breakout/arduino-wiring-test it seems you can communicate over I²C, so it seems that's all the device driver needs an input if I understand correctly.

The LEDs support could be done in a similar way to the apa102 driver API https://periph.io/x/devices/v3/apa102. 

Do not copy arduino or python code as is since the license is incompatible.

different55, 2021-02-19:
What should I be aware of when it comes to copying? Obviously we're not meaning literal copying and pasting so I'm not sure how careful I should be here.

All that's needed for this device is pure I²C, in the Python library Adafruit implemented it so that every peripheral device supported by the Seesaw is broken out into its own module, each inheriting from the main module. The Seesaw makes them all accessible over I²C and handles all the details of actually communicating with and monitoring the hardware connected to it.
-------- Original Message --------

maruel, 2021-02-19:
The guidance is don't copy. Design something new that strictly makes sense with periph's design.

different55, 2021-02-19:
Sounds good, I'll check back in if I have any more questions. Thanks for the help!