periph/devices #11: Adding support for Inky auto-detecting board type

Thereatra, 2021-04-04:
I have been using the Inky device driver for a project and needed the automatic board detection feature of the official Python library. There is no official board documentation so I have simply ported the feature over, from the Python library, for my own project. The code is still rather crude but I would be interested in cleaning it up and adding it to the device driver.
Where should I start?

maruel, 2021-04-04:
Hi! I don't have such devices so I can't help directly.
According to history, you may want to talk with @davidsansome and @hatstand for guidance.

davidsansome, 2021-04-05:
That sounds cool, I'd be happy to review a pull request for board detection! I've got a red/black wHAT I can try it with, and I think @hatstand has some kind of pHAT.

I'm not sure if there's a convention for hardware detection methods elsewhere in Periph, but what about a function in inky.go that returns an Opts struct?

Thereatra, 2021-04-05:
I'm new to both the hardware and go space so I'm not that sure of the conventions, but since the Inky is designed as a Pi hat out of the box I was thinking a function that could return a Dev object with the defaults for pi, would be very useful. This is also what I have implemented currently, but stripping it down if it is out of scope for Periph shouldn't be a problem.

closed 2021-04-12 by Thereatra

maruel, 2021-04-12:
Thanks @Thereatra for doing the change! I still need to tag a new version, will do soon.