periph/devices #15: devices v3.6.11 causes d2xx build failures on WSL2

lutzky, 2021-05-30:
My build uses `devices` (specifically ssd1306) and works fine with v3.6.10. However, upgrading to v3.6.11 adds a dependency of `periph.io/x/d2xx v0.0.1`, which causes build failures on WSL. I'm cross-compiling like so:

```
export CC=arm-linux-gnueabi-gcc CGO_ENABLED=1 GOOS=linux GOARM=6 GOARCH=arm
go build main.go
```

...and get the following errors:

```
# periph.io/x/d2xx
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftd2xx.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftd2xx.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftd2xx-private.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftd2xx-private.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(Event.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(Event.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(usb_control.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(usb_control.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(bulk_in.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(bulk_in.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(device-list.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(device-list.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(idtable.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(idtable.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ee.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ee.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232r.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232r.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232h.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee232h.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee2232.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee2232.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee2232h.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee2232h.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee4232h.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(ftee4232h.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(fteeftx.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(fteeftx.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(serialno.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(serialno.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(core.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(core.o)
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: error: ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(descriptor.o) uses VFP register arguments, $WORK/b164/_cgo_.o does not
/usr/lib/gcc-cross/arm-linux-gnueabi/9/../../../../arm-linux-gnueabi/bin/ld: failed to merge target specific data of file ../go/pkg/mod/periph.io/x/d2xx@v0.0.1/third_party/libftd2xx_linux_arm6hf_v1.4.6.a(descriptor.o)
[... cut ...]
func main() {
  host.Init()
}
```
2. Build it like so:
```
export CC=arm-linux-gnueabi-gcc CGO_ENABLED=1 GOOS=linux GOARM=6 GOARCH=arm
go build main.go
```
3. See errors above.

**Expected behavior**
Should at least *build* successfully, like it did in v3.6.10.

**Platform (please complete the following information):**
 - OS: WSL2 running Ubuntu 20.04.2 LTS
 - Board: Irrelevant, cross-compiling (eventual target is RPi0 fwiw)
 - Device: Irrelevant (eventual targets is ssd1306)

**Additional context**
n/a

maruel, 2021-05-30:
Thanks for the report, will check this ASAP.

maruel, 2021-05-30:
I think it's because you are using a toolchain built with emulated floating point. Can you confirm?