periph/devices #23: Support MPU9250 over I2C

bryanjhv, 2021-11-05:
**What kind of new feature are you looking for?**
- Hardware: Support MPU9250 over I2C, currently only implementation uses SPI which is fine but many mixed chips like the GY-91 expose it as I2C. Another option would be to make `mpu9250.Proto` interface with public methods so one can provide an implementation for it (like `SpiTransport` does).

**Do you plan to:**
- Contribute an initial driver: No
- Write unit tests: No
- Update https://github.com/periph/cmd to use the new functionality: No

bezineb5, 2022-02-06:
I have an implementation of MPU9250 over I2C, with embedded AK8963 I2C magnetometer. However, it's not based upon the code of the existing MPU9250 driver, so I never submitted it (the interfaces are not compatible).
I can share the code if still needed.
@maruel Any interest in this MPU9250 I2C + AK8963? (the AK8963 is actually the magnetometer part of the MPU9250, exposed as a separate device on the I2C bus)

maruel, 2022-02-06:
Sure!

gsexton, 2025-03-16:
Implemented in 

https://github.com/periph/devices/pull/101

closed 2025-03-16 by gsexton