periph/devices #60: dtxx: Implement driver for DHT-11 / DHT-22

maruel, 2023-09-20:
Ported from https://github.com/google/periph/issues/225.

maruel, 2023-09-20:
Requires https://github.com/periph/conn/issues/32.

gsexton, 2025-01-13:
These sensors are problematic. They use a proprietary bitbang protocol. Implementing them would require a device driver because of the strict timing requirements.

Here's a more in-depth review of the associated issues:

https://learn.adafruit.com/modern-replacements-for-dht11-dht22-sensors/why-are-they-probelmatic

Periph has support for AHT20, and AM2320 devices now, which are the next generation components from the vendor. Additionally, support has been added for the TI HDC302X sensor for those desiring a more accurate sensor.

closed 2025-01-13 by gsexton

maruel, 2025-01-17:
My hope was to support these with the big banging protocol, which would be driven by a chip. A good example would be a ftdi FT232H which would be able to fit the timing requirements.